smariel/PTree #10: Efficiency of DC/DC with many points

guillaumezin, 2018-04-04:
Hi,

It would be great to support more than one efficiency point for DC/DC, for instance:
- 90% at 2A
- 70% at 1A
- 65% at 500mA

And to calculate between points using linear interpolation. 4 or 5 points would be enough.

Thanks.

Guillaume

smariel, 2018-05-02:
Coded in 30f3546, available in the next version.
Little form to add a point on the graph.
Click on a point to delete it.
Should be compatible with the older versions.

Rules:
- 1 point by default, 90% @ 1A (= 90% @ any A)
- if no point, 100% @ any A
- if Amps < lowest point, the lowest point is used
- if Amps > highest point, the highst point is used 
- if Amps between two points, linear interpolation (y=ax+b)

[image]

smariel, 2018-05-12:
Available in release [v1.1.0](https://github.com/smariel/PTree/releases/tag/v1.1.0)

closed 2018-05-12 by smariel