smariel/PTree #70: Dropout for LDO

hvanreeth, 2019-09-16:
It would be nice to add a "Dropout" field for LDO converters.
Because dropout is dependent of output current, a chart-like characteristic may be used (same as efficiency for DC/DC).

smariel, 2019-09-16:
The dropout is an interesting parameter.
How exactly would you imagine the implementation ? I think it may be a "forced" limit. For example, if Vdrop=500mV, Vin=3V and Vout=2,8V, the real Vout may be forced to 2,5V with an indicator. But if Vout is set to 2V, Vdrop would be ignore.
What do you think ?

hvanreeth, 2019-09-17:
I see 2 possible implementations : 
- The "force limit" implementation you described.
- A "warning" implementation. In your example (Vin = 3V, Vout = 2.8V, Vdrop = 0.5V), the output voltage of the LDO will stay at 2.8V, but the interface displays a warning. The advantage here is that the LDO "child" items will not be impacted by the dropout issue, and their power will remain nominal.

Maybe the first solution is the best because the most realistic. The dropout could always be ignored if set to 0.

closed 2020-01-11 by smariel