smariel/PTree #72: Input voltage limits

hvanreeth, 2019-09-16:
It would be useful for LDO and DC/DC converters to add "input voltage limits (min and max)" characteristics, and to display a warning (like text in red, or popup) when the input voltage of the converters is out of bounds.

smariel, 2019-09-16:
Very good idea. Not too hard to code.
Will be included in the next major release.
Thanks !

hvanreeth, 2019-09-17:
Great !
When do you plan to publish this major release ?

smariel, 2019-09-17:
I have no idea... there is no roadmap at all because it's all on a my free time.
Minor releases consist in bug corrections, mostly blocking bugs. Or simple enhancements that does not risk to break the software.
This would be in a major release because adding inputs is a always risk and breaks the compatibility with older versions.

But your three proposals are good ideas and doesn't seems too difficult. I'll try to implement them as quick as possible.

closed 2020-01-11 by smariel