smariel/PTree #93: Error popup on 'Help' > 'Equation Summary'

dracorpg, 2021-04-16:
Trying to reach the equation summary shows an error popup instead.

**To Reproduce**
1. Click on 'Help' menu > 'Equation Summary'

**Expected behavior**
The PDF file with equations should be displayed :)

**Screenshots**
I'll dispense with an actual screenshot, it's a relatively bland error popup that reads:
> A JavaScript error occured in the main process
>
> Uncaught Exception:
> TypeError: require(..).shell.openitem is not a function
>   at click ([PTree]\resources\app.asar\main.js:211:41)
>   at Menultem.click (electron/js2c/browser_init.js:73:1683)
>   at Object.a._executeCommand (electron/js2c/browser_init.js:81:2343)

**Desktop (please complete the following information):**
 - OS: Windows (also confirmed on Fedora)
 - Version: 2.0 (seems to be a regression from 1.8)

smariel, 2021-04-17:
Thanks !

`shell.openItem()` is no longer a function since Electron v9.0.0. A fix has to be found to open an external file with the default program. shell.openExternal may be a solution.

For information, on Windows, the PDF file is located in the root folder of the application.

dracorpg, 2021-04-17:
From a cursory glance at the Electron doc, [`shell.openPath()`](https://www.electronjs.org/docs/api/shell#shellopenpathpath) looks to me like the most direct replacement for `shell.openItem()`
... and actually, 9.0 API breaking change notes say it's precisely the case: https://www.electronjs.org/blog/electron-9-0#api-changes :)

commit 71619af6 (2021-04-19): open equation menu fix #93
closed 2021-04-19 by smariel