smariel/PTree #99: buck regulator current sharing

Poupsfamily, 2021-05-03:
Hello Sylvain,

When we are doing power supply architecture, at board level, sometimes we need to make current sharing with 2 buck because one buck is out of derating on its output current.
It could be interesting to be able to do this with Ptree.

The attached file show what I actually do in excel file. You will see current shring on the PG3V3

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

smariel, 2021-05-03:
Hi,
Thank you for this proposal. This is a great idea but a very complex integration. I will think about it.

dracorpg, 2021-06-06:
Hello to both of you,

I believe an useful implementation could be as simple as a "paralleled amount" attribute for DC/DC & LDO regulators (possibly shown in some explicit way on the schematic, like a "x2" badge in a corner or using a "multiple-Z-stacked blocks" effect). This would assume perfect current sharing between each one of the N paralleled instances so that no advanced computation is required (loss-curve curve & current limit logic would use 1/N of the load current, then total regulator loss = unit loss x N).

What do you think?